OpenXiangShan/GEM5 #41: Failed to run hello world

HarsonLau, 2023-05-25:
I compiled the code from the branch `dbp-merge-xsdev-221010`.
```bash
git switch dbp-merge-xsdev-221010

scons build/RISCV/gem5.debug -j16  
```
And tried to run a hello world program.
```bash
./build/RISCV/gem5.debug configs/example/se.py --cpu-type=DerivO3CPU --caches -c tests/test-progs/hello/bin/riscv/linux/hello
```

It failed with 
```txt
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version [DEVELOP-FOR-22.1]
gem5 compiled May 25 2023 12:01:17
gem5 started May 25 2023 12:24:53
gem5 executing on mprc2, pid 1587869
command line: ./build/RISCV/gem5.debug configs/example/se.py --cpu-type=DerivO3CPU --caches -c tests/test-progs/hello/bin/riscv/linux/hello

build/RISCV/base/loader/image_file_data.cc:107: info: Loading file tests/test-progs/hello/bin/riscv/linux/hello
build/RISCV/base/loader/image_file_data.cc:127: info: File size is 4814352 bytes
build/RISCV/base/loader/image_file_data.cc:133: info: First 4 bytes are 0x7f 0x45 0x4c 0x46
build/RISCV/base/loader/image_file_data.cc:135: info: Mapped start address is ELF, 0x7ffff4d3e000
Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
build/RISCV/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (512 Mbytes)
build/RISCV/base/loader/image_file_data.cc:107: info: Loading file tests/test-progs/hello/bin/riscv/linux/hello
build/RISCV/base/loader/image_file_data.cc:127: info: File size is 4814352 bytes
build/RISCV/base/loader/image_file_data.cc:133: info: First 4 bytes are 0x7f 0x45 0x4c 0x46
build/RISCV/base/loader/image_file_data.cc:135: info: Mapped start address is ELF, 0x7fffd4cfe000
build/RISCV/cpu/o3/cpu.cc:332: warn: Difftest is disabled
0: system.remote_gdb: listening for remote gdb on port 7000
**** REAL SIMULATION ****
build/RISCV/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
gem5.debug: build/RISCV/cpu/pred/decoupled_bpred.cc:155: std::pair<bool, bool> gem5::branch_prediction::DecoupledBPU::decoupledPredict(const StaticInstPtr&, const InstSeqNum&, gem5::PCStateBase&, gem5::ThreadID): Assertion `pc.instAddr() < end && pc.instAddr() >= start' failed.
```

I'm wondering if the way I run hello world is wrong or the branch I'm using has a bug？

shinezyy, 2023-05-25:
Feeling sorry for the broken project.

1. The working version of calibrated GEM5 (against XS) has not been released. But it will be released soon, maybe in 1~2 weeks.
2. In our internal version, we currently only support FS mode, and se.py is also broken.

HarsonLau, 2023-05-25:
Looking forward to the release of the working version. Thanks!

closed 2023-05-25 by HarsonLau

shinezyy, 2023-05-31:
We make it publicly available just now. Although the SE mode works, we do not suggest to use it for performance evaluation.
Because under SE mode,
- difftest is not supported
- the performance has not been calibrated
- the performance of system call and context switch are not modeled
- TLB and PTW cannot be properly evaluated
- the performance of hardware prefetcher will be skewed: VA-based
algorithm might not work, while PA-based algorithm will be favored
because of flatten address space.