OpenXiangShan/GEM5 #46: gem5 怎么跑coremark

qqli579, 2024-01-25:
用你们提供的配置参数跑回报错误。参数如下：
```
                "$gem5_home/build/RISCV/gem5.fast \
                    $gem5_home/configs/example/fs.py\
                    --xiangshan-system \
                    --bare-metal \
                    --cpu-type=DerivO3CPU \
                    --mem-size=8GB \
                    --caches --cacheline_size=64 \
                    --l1i_size=64kB --l1i_assoc=8 \
                    --l1d_size=64kB --l1d_assoc=8 \
                    --l1d-hwp-type=XSCompositePrefetcher \
                    --short-stride-thres=0 \
                    --l2cache --l2_size=1MB --l2_assoc=8 \
                    --l3cache --l3_size=16MB --l3_assoc=16 \
                    --l1-to-l2-pf-hint \
                    --l2-hwp-type=WorkerPrefetcher \
                    --l2-to-l3-pf-hint \
                    --l3-hwp-type=WorkerPrefetcher \
                    --mem-type=DRAMsim3 \
                    --dramsim3-ini=$gem5_home/ext/dramsim3/xiangshan_configs/xiangshan_DDR4_8Gb_x8_3200_2ch.ini \
                    --bp-type=DecoupledBPUWithFTB --enable-loop-predictor \
                    --kernel=./benchmark/image/coremark.2timesriscv \
                    --command-line='0x0 0x0 0x66 0 7 1 2000' |& tee run.log" Enter
```
报下面的错误
```
Global frequency set at 1000000000000 ticks per second
WARNING: Output directory ext/dramsim3/DRAMsim3/ not exists! Using current directory for output!
fatal: system.workload.bootloader without default or user set value
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version [DEVELOP-FOR-22.1]
gem5 compiled Jan 25 2024 10:35:49
gem5 started Jan 25 2024 10:36:29
gem5 executing on n168-020-004, pid 17185
command line: ./GEM5-xs-dev/build/RISCV/gem5.fast ./GEM5-xs-dev/configs/example/fs.py --xiangshan-system --cpu-type=DerivO3CPU --mem-size=8GB --caches --cacheline_size=64 --l1i_size=64kB --l1i_assoc=8 --l1d_size=64kB --l1d_assoc=8 --l1d-hwp-type=XSCompositePrefetcher --short-stride-thres=0 --l2cache --l2_size=1MB --l2_assoc=8 --l3cache --l3_size=16MB --l3_assoc=16 --l1-to-l2-pf-hint --l2-hwp-type=WorkerPrefetcher --l2-to-l3-pf-hint --l3-hwp-type=WorkerPrefetcher --mem-type=DRAMsim3 --dramsim3-ini=./GEM5-xs-dev/ext/dramsim3/xiangshan_configs/xiangshan_DDR4_8Gb_x8_3200_2ch.ini --bp-type=DecoupledBPUWithFTB --enable-loop-predictor --kernel=./benchmark/image/coremark.2timesriscv '--command-line=0x0 0x0 0x66 0 7 1 2000'

[<m5.params.AddrRange object at 0x7f42d71dfc10>]
['basic']
db_switches: []
Attach 1 decoders to thread with addr: <orphan System>.cpu.decoder
Create threads for test sys cpu (RiscvO3CPU)
Add dtb for L1D prefetcher
Add L2 prefetcher as downstream of L1D prefetcher
Add L3 prefetcher as downstream of L2 prefetcher
Add dtb for L2 prefetcher
Finish memory system configuration
No cpu_class provided
```

shinezyy, 2024-01-25:
https://github.com/OpenXiangShan/GEM5/blob/6a85ffb479c8c1ecff432ddd5801216b4e829580/util/warmup_scripts/simple_gem5.sh#L74C1-L77C7

shinezyy, 2024-01-25:
你用的是fs.py的，但是参数风格是se.py的参数风格，你在尝试从从gem5的命令行给应用传参数。
但是，fs不支持从gem5的命令行给应用传参数。

shinezyy, 2024-01-25:
关于“gem5 怎么跑coremark ”，本项目目前只支持和香山处理器一样的方式运行：

1. 要么是不带参数的bare metal core mark，用这里的方法运行：
> https://github.com/OpenXiangShan/GEM5/blob/6a85ffb479c8c1ecff432ddd5801216b4e829580/util/warmup_scripts/simple_gem5.sh#L74C1-L77C7

3. 要么生成香山自定义的全系统仿真的checkpoint，如何生成请看文档：https://github.com/OpenXiangShan/GEM5?tab=readme-ov-file#produce-rvgcpt-checkpoints-with-nemu

closed 2024-01-25 by shinezyy