OpenXiangShan/GEM5 #47: gem5 has encountered a segmentation fault

qqli579, 2024-01-30:
simple_gem5.sh 跑baremetal bin 的时候报这个错误
```
Attach 1 decoders to thread with addr: <orphan System>.cpu.decoder
Create threads for test sys cpu (RiscvO3CPU)
Add dtb for L1D prefetcher
Add L2 prefetcher as downstream of L1D prefetcher
Add L3 prefetcher as downstream of L2 prefetcher
Add dtb for L2 prefetcher
Finish memory system configuration
No cpu_class provided
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.bop_large
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.bop_small
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.ipcp
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.spp
Registering probe listeners for Prefetcher system.l2.prefetcher
Registering probe listeners for Prefetcher system.l3.prefetcher
**** REAL SIMULATION ****
build/RISCV/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
gem5 has encountered a segmentation fault!
```
重复步骤：
按照simple_gem5.sh 修改脚本如下：
```
# DO NOT track your local updates in this script!
# set -x

export gem5_home=/xiangshang/gem5/GEM5-xs-dev # The root of GEM5 project
export gem5=$gem5_home/build/RISCV/gem5.fast # GEM5 executable

# Note 1: workload list contains the workload name, checkpoint path, and parameters, looks like:
#       astar_biglakes_122060000000 astar_biglakes_122060000000_0.244818/0/ 0 0 20 20
#       bwaves_1003220000000 bwaves_1003220000000_0.036592/0/ 0 0 20 20
# Note 2: The meaning of fields:
# workload_name, checkpoint_path, skip insts(usually 0), functional_warmup insts(usually 0), detailed_warmup insts (usually 20), sample insts
# Note 3: you can write a script to generate such a list accordingly
export desc_dir=/xiangshang/benchmark/coremark/fs
export workload_list=/xiangshang/benchmark/coremark/fs/int_list.lst

# The checkpoint directory. We will find checkpoint_path in workload_list
# under this directory to get the checkpoint path.
export cpt_dir='/xiangshang/benchmark/image'

# A tag to identify current batch run
export tag="an-example-to-run-gem5-with-composite-prefetcher"

export log_file='log.txt'

export ds=$(pwd)  # data storage. It is specific for BOSC machines, you can ignore it

export top_work_dir=$tag
export full_work_dir=$ds/exec-storage/$top_work_dir  # work dir wheter stats data stored

mkdir -p $full_work_dir
ln -sf $full_work_dir .  # optional, you can customize it yourself

check() {
    if [ $1 -ne 0 ]; then
        echo FAIL
        touch abort
        exit
    fi
}

function run() {
    set -x
    cpt=$1
    dw_len=${2:-20000000}
    # dw_len=${2:-1525605}
    total_detail_len=${3:-40000000}
    if [[ -n "$4" ]]; then
        work_dir=$4
    else
        work_dir=$PWD
    fi
    arch_db=${5:-0}

    cd $work_dir

    if test -f "completed"; then
        echo "Already completed; skip $cpt"
        return
    fi

    rm -f abort
    rm -f completed

    cpt_name=$(basename -- "$cpt")
    extension="${cpt_name##*.}"

    # replace the path of gcpt.bin with your gcpt restorer
    # gcpt restorer can be found in https://github.com/OpenXiangShan/NEMU/tree/gem5-ref-main/resource/gcpt_restore
    # Please use gem5-ref-main branch
    cpt_option="--generic-rv-cpt=$cpt --gcpt-restorer=/xiangshang/NEMU-gem5-ref-main/resource/gcpt_restore/build/gcpt.bin"

    # You can also pass a baremetal bin here
    if [ $extension != "gz" ]; then
        cpt_option="--generic-rv-cpt=./xiangshang/benchmark/image/coremark.bare.1timesriscv --raw-cpt"
    fi

    if [[ "$arch_db" -eq "0" ]]; then
        arch_db_args=
    else
        arch_db_args="--enable-arch-db --arch-db-file=mem_trace.db --arch-db-fromstart=True"
    fi

    if [[ -z "$crash_tick" ]]; then
        crash_tick=-1
    fi
    if [[ -z "$capture_cycles" ]]; then
        capture_cycles=30000
    fi
    start=$(($crash_tick - 500*$capture_cycles))
    # start=$crash_tick
    start=$(($start>0 ? $start : 0))
    end=$(($crash_tick + 500*$capture_cycles))
    start_end=" --debug-start=$start --debug-end=$end "
    if [[ -n "$debug_flags" ]]; then
        debug_flag_args=" --debug-flag=$debug_flags "
    else
        echo "No debug flag set"
        debug_flag_args=
        start_end=
    fi
    # --debug-flags=CommitTrace \

    if [[ $crash_tick = -1 ]]; then
        start_end=
        debug_flag_args=
    fi

    echo "total_detail_len: $total_detail_len"
    # gdb -ex run --args \

    # Note 1: Use DecoupledBPUWithFTB to enable nanhu's decoupled frontend
    # Note 2: MUST use DRAMsim3, or performance is skewed
    # To enable DRAMSim3, follow ext/dramsim3/README
    # Note 3: By default use MultiPrefetcher (SMS + BOP) as L2 prefetcher
    # Note 4: Recommend to enable Difftest
    ######## Some additional args:
    $gem5 $debug_flag_args $start_end \
        $gem5_home/configs/example/fs.py \
        --xiangshan-system --cpu-type=DerivO3CPU \
        --mem-size=8GB \
        --caches --cacheline_size=64 \
        --l1i_size=64kB --l1i_assoc=8 \
        --l1d_size=64kB --l1d_assoc=8 \
        --l1d-hwp-type=XSCompositePrefetcher \
        --short-stride-thres=0 \
        --l2cache --l2_size=1MB --l2_assoc=8 \
        --l3cache --l3_size=16MB --l3_assoc=16 \
        --l1-to-l2-pf-hint \
        --l2-hwp-type=WorkerPrefetcher \
        --l2-to-l3-pf-hint \
        --l3-hwp-type=WorkerPrefetcher \
        --mem-type=DRAMsim3 \
        --dramsim3-ini=$gem5_home/ext/dramsim3/xiangshan_configs/xiangshan_DDR4_8Gb_x8_3200_2ch.ini \
        --bp-type=DecoupledBPUWithFTB --enable-loop-predictor \
        --enable-difftest \
        $arch_db_args $cpt_option \
        --maxinsts=3849417830
    check $?

    # Here is a scratchpad for frequently used options

        # Enable complex stride component or SPP component in composite prefetcher
        # --l1d-enable-cplx \
        # --l1d-enable-spp \

        # Record arch db traces only after warmup
        #  --arch-db-fromstart=False 

        # Enable loop predictor and loop buffer
        # --enable-loop-predictor \
        # --enable-loop-buffer \

        # Employ an ideal L2 cache with nearly-perfetch hit rate and low-access latency
        # --mem-type=SimpleMemory \
        # --ideal-cache \

    # Debugging memory corruption or memory leak
    # valgrind -s --track-origins=yes --leak-check=full --show-leak-kinds=all --log-file=valgrind-out-2.txt --error-limit=no -v \

    touch completed
}

function prepare_env() {
    set -x
    echo "prepare_env $@"
    all_args=("$@")
    task=${all_args[0]}
    task_path=${all_args[1]}
    gz=$(find -L $cpt_dir -wholename "*${task_path}*gz" | head -n 1)
    echo $gz
    work_dir=$top_work_dir/$task
    echo $work_dir
    mkdir -p $work_dir
}

function arg_wrapper() {
    prepare_env $@

    all_args=("$@")
    args=(${all_args[0]})

    k=1000
    M=$((1000 * $k))

    skip=${args[2]}
    fw=${args[3]}
    dw=${args[4]}
    sample=${args[5]}

    total_M=$(( ($dw + $sample)*$M ))
    dw_M=$(( $dw*$M ))

    run $gz $dw_M $total_M $work_dir 0 >$work_dir/$log_file 2>&1
}

function single_run() {
    # run /nfs-nvme/home/zhouyaoyang/projects/nexus-am/apps/cachetest_i/build/cachetest_i-riscv64-xs.bin
    task=$tag
    work_dir=$full_work_dir
    mkdir -p $work_dir

    # Note: If you are debugging with single run, following 3 variables are mandatory.
    # - It prints debug info in tick range: (crash_tick - 500 * capture_cycles, crash_tick + 500 * capture_cycles)
    # - If you want to print debug info from beginning, set crash_tick to 0, and set capture_cycles to a large number

    # crash_tick=$(( 0 ))
    # capture_cycles=$(( 250000 ))
    # debug_flags=CommitTrace  # If you unset debug_flags, no debug print will be there

    # If you unset debug_flags or crash_tick, no debug print will be there
    # Common used flags for debug/tuning
    # debug_flags=CommitTrace,IEW,Fetch,LSQUnit,Cache,Commit,IQ,LSQ,PageTableWalker,TLB,MSHR
    warmup_inst=$(( 20 * 10**6 ))
    max_inst=$(( 40 * 10**6 ))

    # debug_gz=/nfs-nvme/home/share/checkpoints_profiles/spec06_rv64gcb_o2_20m/take_cpt/mcf_191500000000_0.105600/0/_191500000000_.gz
    debug_gz=/nfs-nvme/home/share/checkpoints_profiles/spec06_rv64gcb_o2_20m/take_cpt/libquantum_1006500000000_0.149838/0/_1006500000000_.gz
    rm -f $work_dir/completed
    rm -f $work_dir/abort
    run $debug_gz $warmup_inst $max_inst $work_dir 1 > $work_dir/$log_file 2>&1
}

export -f check
export -f run
export -f single_run
export -f arg_wrapper
export -f prepare_env

function parallel_run() {
    # We use gnu parallel to control the parallelism.
    # If your server has 32 core and 64 SMT threads, we suggest to run with no more than 32 threads.
    export num_threads=1
    cat $workload_list | parallel -a - -j $num_threads arg_wrapper {}
}

# Usually, I use paralell run to benchmark, and use single_run to debug
parallel_run
# single_run
```

然后执行：./simple_gem5.sh

shinezyy, 2024-01-30:
能否提供log.txt？

qqli579, 2024-01-30:
```
+ set -x
+ cpt=102000000
+ dw_len=10102000000
+ total_detail_len=an-example-to-run-gem5-with-composite-prefetcher/coremark
+ [[ -n 0 ]]
+ work_dir=0
+ arch_db=0
+ cd 0
environment: line 10: cd: 0: No such file or directory
+ test -f completed
+ rm -f abort
+ rm -f completed
++ basename -- 102000000
+ cpt_name=102000000
+ extension=102000000
+ cpt_option='--generic-rv-cpt=102000000 --gcpt-restorer=/xiangshang/NEMU-gem5-ref-main/resource/gcpt_restore/build/gcpt.bin'
+ '[' 102000000 '!=' gz ']'
+ cpt_option='--generic-rv-cpt=/xiangshang/benchmark/image/coremark.bare.1timesriscv --raw-cpt'
+ [[ 0 -eq 0 ]]
+ arch_db_args=
+ [[ -z '' ]]
+ crash_tick=-1
+ [[ -z '' ]]
+ capture_cycles=30000
+ start=-15000001
+ start=0
+ end=14999999
+ start_end=' --debug-start=0 --debug-end=14999999 '
+ [[ -n '' ]]
+ echo 'No debug flag set'
No debug flag set
+ debug_flag_args=
+ start_end=
+ [[ -1 = -1 ]]
+ start_end=
+ debug_flag_args=
+ echo 'total_detail_len: an-example-to-run-gem5-with-composite-prefetcher/coremark'
total_detail_len: an-example-to-run-gem5-with-composite-prefetcher/coremark
+ /xiangshang/gem5/GEM5-xs-dev/build/RISCV/gem5.fast /xiangshang/gem5/GEM5-xs-dev/configs/example/fs.py --xiangshan-system --cpu-type=DerivO3CPU --mem-size=8GB --caches --cacheline_size=64 --l1i_size=64kB --l1i_assoc=8 --l1d_size=64kB --l1d_assoc=8 --l1d-hwp-type=XSCompositePrefetcher --short-stride-thres=0 --l2cache --l2_size=1MB --l2_assoc=8 --l3cache --l3_size=16MB --l3_assoc=16 --l1-to-l2-pf-hint --l2-hwp-type=WorkerPrefetcher --l2-to-l3-pf-hint --l3-hwp-type=WorkerPrefetcher --mem-type=DRAMsim3 --dramsim3-ini=/xiangshang/gem5/GEM5-xs-dev/ext/dramsim3/xiangshan_configs/xiangshan_DDR4_8Gb_x8_3200_2ch.ini --bp-type=DecoupledBPUWithFTB --enable-loop-predictor --enable-difftest --generic-rv-cpt=/xiangshang/benchmark/image/coremark.bare.1timesriscv --raw-cpt --maxinsts=3849417830
Global frequency set at 1000000000000 ticks per second
WARNING: Output directory ext/dramsim3/DRAMsim3/ not exists! Using current directory for output!
build/RISCV/arch/riscv/bare_metal/fs_workload.cc:60: info: No bootload provided, because using XS GCPT, reset to 0x80000000
build/RISCV/cpu/base.cc:223: warn: Difftest is disabled
build/RISCV/base/statistics.hh:280: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
111
build/RISCV/base/remote_gdb.cc:381: warn: Sockets disabled, not accepting gdb connections
build/RISCV/mem/physical.cc:526: info: copying /xiangshang/benchmark/image/coremark.bare.1timesriscv to pmem 0x7fb8ff800000
build/RISCV/mem/physical.cc:544: info: First 4 bytes are 0x7f 0x45 0x4c 0x46
build/RISCV/sim/system.cc:556: info: Restoring from Xiangshan RISC-V Checkpoint
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version [DEVELOP-FOR-22.1]
gem5 compiled Jan 25 2024 10:35:49
gem5 started Jan 30 2024 16:14:48
gem5 executing on n168-020-004, pid 3272475
command line: /xiangshang/gem5/GEM5-xs-dev/build/RISCV/gem5.fast /xiangshang/gem5/GEM5-xs-dev/configs/example/fs.py --xiangshan-system --cpu-type=DerivO3CPU --mem-size=8GB --caches --cacheline_size=64 --l1i_size=64kB --l1i_assoc=8 --l1d_size=64kB --l1d_assoc=8 --l1d-hwp-type=XSCompositePrefetcher --short-stride-thres=0 --l2cache --l2_size=1MB --l2_assoc=8 --l3cache --l3_size=16MB --l3_assoc=16 --l1-to-l2-pf-hint --l2-hwp-type=WorkerPrefetcher --l2-to-l3-pf-hint --l3-hwp-type=WorkerPrefetcher --mem-type=DRAMsim3 --dramsim3-ini=/xiangshang/gem5/GEM5-xs-dev/ext/dramsim3/xiangshan_configs/xiangshan_DDR4_8Gb_x8_3200_2ch.ini --bp-type=DecoupledBPUWithFTB --enable-loop-predictor --enable-difftest --generic-rv-cpt=/xiangshang/benchmark/image/coremark.bare.1timesriscv --raw-cpt --maxinsts=3849417830

info: Standard input is not a terminal, disabling listeners.
[<m5.params.AddrRange object at 0x7fbaffdf4130>]
Using raw bbl None
['basic']
db_switches: []
Attach 1 decoders to thread with addr: <orphan System>.cpu.decoder
Create threads for test sys cpu (RiscvO3CPU)
Add dtb for L1D prefetcher
Add L2 prefetcher as downstream of L1D prefetcher
Add L3 prefetcher as downstream of L2 prefetcher
Add dtb for L2 prefetcher
Finish memory system configuration
No cpu_class provided
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.bop_large
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.bop_small
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.ipcp
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.spp
Registering probe listeners for Prefetcher system.l2.prefetcher
Registering probe listeners for Prefetcher system.l3.prefetcher
**** REAL SIMULATION ****
build/RISCV/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
gem5 has encountered a segmentation fault!

+ check 139
+ '[' 139 -ne 0 ']'
+ echo FAIL
FAIL
+ touch abort
+ exit
```

shinezyy, 2024-01-30:
对脚本进行如下修改，然后重新运行

加上gdb：
++ gdb -ex run --args \
$gem5 $debug_flag_args $start_end \
        $gem5_home/configs/example/fs.py \
...

修改single_run 函数：
function single_run() {
...

--    run $debug_gz $warmup_inst $max_inst $work_dir 1 > $work_dir/$log_file 2>&1
++    run $debug_gz $warmup_inst $max_inst $work_dir 1
}
记得在single_run 函数中传入coremark.bin

...

改为调用single_run：
++ # parallel_run
++ single_run

qqli579, 2024-01-30:
在single_run 函数中怎么传入coremark.bin

shinezyy, 2024-01-30:
> 在single_run 函数中怎么传入coremark.bin

debug_gz=/nfs-nvme/home/share/checkpoints_profiles/spec06_rv64gcb_o2_20m/take_cpt/libquantum_1006500000000_0.149838/0/_1006500000000_.gz
 改为

debug_gz=./xiangshang/benchmark/image/coremark.bare.1timesriscv

qqli579, 2024-01-30:
coremark.bin 后面跟的参数‘0x0 0x0 0x66 20000 7 1 2000’ 怎么传入进去

shinezyy, 2024-01-30:
> coremark.bin 后面跟的参数‘0x0 0x0 0x66 20000 7 1 2000’ 怎么传入进去

不支持传参

qqli579, 2024-01-30:
又报这个错误
```
Global frequency set at 1000000000000 ticks per second
build/RISCV/sim/arch_db.cc:37: info: Table created: CREATE TABLE L1MissTrace(ID INTEGER PRIMARY KEY AUTOINCREMENT,PC INT NOT NULL,SOURCE INT NOT NULL,PADDR INT NOT NULL,VADDR INT NOT NULL,STAMP INT NOT NULL,SITE TEXT);
build/RISCV/sim/arch_db.cc:37: info: Table created: CREATE TABLE L1EvictTrace(ID INTEGER PRIMARY KEY AUTOINCREMENT,PADDR INT NOT NULL,STAMP INT NOT NULL,SITE TEXT);
build/RISCV/sim/arch_db.cc:37: info: Table created: CREATE TABLE MemTrace(ID INTEGER PRIMARY KEY AUTOINCREMENT,Tick INT NOT NULL,IsLoad BOOL NOT NULL,PC INT NOT NULL,VADDR INT NOT NULL,PADDR INT NOT NULL,Issued INT NOT NULL,Translated INT NOT NULL,Completed INT NOT NULL,Committed INT NOT NULL,Writenback INT NOT NULL,PFSrc INT NOT NULL);
WARNING: Output directory ext/dramsim3/DRAMsim3/ not exists! Using current directory for output!
build/RISCV/arch/riscv/bare_metal/fs_workload.cc:60: info: No bootload provided, because using XS GCPT, reset to 0x80000000
build/RISCV/cpu/base.cc:223: warn: Difftest is disabled
build/RISCV/base/statistics.hh:280: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
111
0: system.remote_gdb: listening for remote gdb on port 7000
build/RISCV/mem/physical.cc:526: info: copying /xiangshang/benchmark/image/coremark.1timesriscv to pmem 0x7f45db200000
build/RISCV/mem/physical.cc:544: info: First 4 bytes are 0x7f 0x45 0x4c 0x46
build/RISCV/sim/system.cc:556: info: Restoring from Xiangshan RISC-V Checkpoint
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version [DEVELOP-FOR-22.1]
gem5 compiled Jan 30 2024 16:33:48
gem5 started Jan 30 2024 17:12:34
gem5 executing on n168-020-004, pid 88390
command line: /xiangshang/gem5/GEM5-xs-dev/build/RISCV/gem5.opt /xiangshang/gem5/GEM5-xs-dev/configs/example/fs.py --xiangshan-system --cpu-type=DerivO3CPU --mem-size=8GB --caches --cacheline_size=64 --l1i_size=64kB --l1i_assoc=8 --l1d_size=64kB --l1d_assoc=8 --l1d-hwp-type=XSCompositePrefetcher --short-stride-thres=0 --l2cache --l2_size=1MB --l2_assoc=8 --l3cache --l3_size=16MB --l3_assoc=16 --l1-to-l2-pf-hint --l2-hwp-type=WorkerPrefetcher --l2-to-l3-pf-hint --l3-hwp-type=WorkerPrefetcher --mem-type=DRAMsim3 --dramsim3-ini=/xiangshang/gem5/GEM5-xs-dev/ext/dramsim3/xiangshan_configs/xiangshan_DDR4_8Gb_x8_3200_2ch.ini --bp-type=DecoupledBPUWithFTB --enable-loop-predictor --enable-difftest --enable-arch-db --arch-db-file=mem_trace.db --arch-db-fromstart=True --generic-rv-cpt=/xiangshang/benchmark/image/coremark.1timesriscv --raw-cpt --warmup-insts-no-switch=20000000 --maxinsts=40000000

[<m5.params.AddrRange object at 0x7f47db60a350>]
Using raw bbl None
['basic']
db_switches: []
Attach 1 decoders to thread with addr: <orphan System>.cpu.decoder
Create threads for test sys cpu (RiscvO3CPU)
Add dtb for L1D prefetcher
Add L2 prefetcher as downstream of L1D prefetcher
Add L3 prefetcher as downstream of L2 prefetcher
Add dtb for L2 prefetcher
Finish memory system configuration
No cpu_class provided
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.bop_large
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.bop_small
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.ipcp
Registering probe listeners for Prefetcher system.cpu.dcache.prefetcher.spp
Registering probe listeners for Prefetcher system.l2.prefetcher
Registering probe listeners for Prefetcher system.l3.prefetcher
**** REAL SIMULATION ****
build/RISCV/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
gem5.opt: build/RISCV/cpu/pred/ftb/decoupled_bpred.cc:1194: void gem5::branch_prediction::ftb_pred::DecoupledBPUWithFTB::trapSquash(unsigned int, unsigned int, gem5::Addr, const gem5::PCStateBase&, gem5::ThreadID, const unsigned int&): Assertion `it != fetchStreamQueue.end()' failed.
Program aborted at tick 69930
```
完整log如下：
```
export gem5_home=/mnt/project/hukun/xiangshang/gem5/GEM5-xs-dev
+ gem5_home=/mnt/project/hukun/xiangshang/gem5/GEM5-xs-dev
+ export gem5=/mnt/project/hukun/xiangshang/gem5/GEM5-xs-dev/build/RISCV/gem5.opt
+ gem5=/mnt/project/hukun/xiangshang/gem5/GEM5-xs-dev/build/RISCV/gem5.opt
+ export desc_dir=/projects/BatchTaskTemplate/resources/simpoint_cpt_desc
+ desc_dir=/projects/BatchTaskTemplate/resources/simpoint_cpt_desc
+ export workload_list=/projects/BatchTaskTemplate/resources/simpoint_cpt_desc/spec06_rv64gcb_o2_20m__cover1.00_top100-normal-0-0-20-20.lst
+ workload_list=/projects/BatchTaskTemplate/resources/simpoint_cpt_desc/spec06_rv64gcb_o2_20m__cover1.00_top100-normal-0-0-20-20.lst
+ export cpt_dir=/nfs-nvme/home/share/checkpoints_profiles/spec06_rv64gcb_o3_20m_gcc12-fpcontr-off/take_cpt
+ cpt_dir=/nfs-nvme/home/share/checkpoints_profiles/spec06_rv64gcb_o3_20m_gcc12-fpcontr-off/take_cpt
+ export tag=an-example-to-run-gem5-with-composite-prefetcher
+ tag=an-example-to-run-gem5-with-composite-prefetcher
+ export log_file=log.txt
+ log_file=log.txt
++ pwd
+ export ds=/mnt/project/hukun/xiangshang/benchmark/stream/fs
+ ds=/mnt/project/hukun/xiangshang/benchmark/stream/fs
+ export top_work_dir=an-example-to-run-gem5-with-composite-prefetcher
+ top_work_dir=an-example-to-run-gem5-with-composite-prefetcher
[... cut ...]
Registering probe listeners for Prefetcher system.l3.prefetcher
**** REAL SIMULATION ****
build/RISCV/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
gem5.opt: build/RISCV/cpu/pred/ftb/decoupled_bpred.cc:1194: void gem5::branch_prediction::ftb_pred::DecoupledBPUWithFTB::trapSquash(unsigned int, unsigned int, gem5::Addr, const gem5::PCStateBase&, gem5::ThreadID, const unsigned int&): Assertion `it != fetchStreamQueue.end()' failed.
Program aborted at tick 69930
+ check 0
+ '[' 0 -ne 0 ']'
+ touch completed

```

shinezyy, 2024-01-30:
请问你可以介绍一下你的coremark.1timesriscv是如何生成的吗？

shinezyy, 2024-01-30:
XS-GEM5只支持香山的workload，比如香山的GCPT或者AM裸机应用。你如果用的是boom的，或者别的，我们就不支持

qqli579, 2024-01-30:
https://github.com/riscv-boom/riscv-coremark/blob/master/riscv64/core_portme.mak
用这个编译的

tastynoob, 2024-01-30:
https://github.com/OpenXiangShan/nexus-am
请参考这个项目进行构建支持香山的AM裸机应用

shinezyy, 2024-01-30:
> https://github.com/riscv-boom/riscv-coremark/blob/master/riscv64/core_portme.mak 用这个编译的

你为什么觉得boom的裸机应用可以在XS-GEM5上运行起来？

closed 2024-03-06 by shinezyy