OpenXiangShan/GEM5 #49: 请问下当前Gem5最新的代码是和nanhu哪个版本对齐的？

DimanChauncey, 2024-03-23:
了解到nanhu有好几个版本，想请问下模拟器具体是和哪个版本进行对齐的，或者对齐的进度。nanhu-G？nanhu v3？

shinezyy, 2024-03-29:
现在的版本没有和南湖对齐了，和昆明湖更接近

sususjysjy, 2024-04-25:
> 现在的版本没有和南湖对齐了，和昆明湖更接近

当前gem5中bpu部分和nanhu同步吗？

eastonman, 2024-04-25:
目前的对齐工作都是朝向昆明湖对齐的，BPU部分也是一样。
RTL的BPU设计在南湖到昆明湖的变化不是很大，因此和南湖也有一定相似程度。

kevinSun20, 2024-04-25:
> 目前的对齐工作都是朝向昆明湖对齐的，BPU部分也是一样。 RTL的BPU设计在南湖到昆明湖的变化不是很大，因此和南湖也有一定相似程度。

感谢回复，那我使用该仓库下gem5得到bpu的性能作为南湖bpu性能，这种做法是不是可取呢？

shinezyy, 2024-04-25:
> > 目前的对齐工作都是朝向昆明湖对齐的，BPU部分也是一样。 RTL的BPU设计在南湖到昆明湖的变化不是很大，因此和南湖也有一定相似程度。
> 
> 感谢回复，那我使用该仓库下gem5得到bpu的性能作为南湖bpu性能，这种做法是不是可取呢？

你可以认为是一个近似，但是我们无法做出承诺。

kevinSun20, 2024-04-25:
> > > 目前的对齐工作都是朝向昆明湖对齐的，BPU部分也是一样。 RTL的BPU设计在南湖到昆明湖的变化不是很大，因此和南湖也有一定相似程度。
> > 
> > 
> > 感谢回复，那我使用该仓库下gem5得到bpu的性能作为南湖bpu性能，这种做法是不是可取呢？
> 
> 你可以认为是一个近似，但是我们无法做出承诺。

ok，感谢你们的回复，我想再问最后一个问题，我看到报告中雁栖湖bpu参考的是UC Berkeley Sonic BOOM，请问南湖bpu参考的是什么呢？

shinezyy, 2024-04-25:
> > > > 目前的对齐工作都是朝向昆明湖对齐的，BPU部分也是一样。 RTL的BPU设计在南湖到昆明湖的变化不是很大，因此和南湖也有一定相似程度。
> > > 
> > > 
> > > 感谢回复，那我使用该仓库下gem5得到bpu的性能作为南湖bpu性能，这种做法是不是可取呢？
> > 
> > 
> > 你可以认为是一个近似，但是我们无法做出承诺。
> 
> ok，感谢你们的回复，我想再问最后一个问题，我看到报告中雁栖湖bpu参考的是UC Berkeley Sonic BOOM，请问南湖bpu参考的是什么呢？

为什么一定要参考谁呢，我们可以独立设计。

kevinSun20, 2024-04-25:
> > > > > 目前的对齐工作都是朝向昆明湖对齐的，BPU部分也是一样。 RTL的BPU设计在南湖到昆明湖的变化不是很大，因此和南湖也有一定相似程度。
> > > > 
> > > > 
> > > > 感谢回复，那我使用该仓库下gem5得到bpu的性能作为南湖bpu性能，这种做法是不是可取呢？
> > > 
> > > 
> > > 你可以认为是一个近似，但是我们无法做出承诺。
> > 
> > 
> > ok，感谢你们的回复，我想再问最后一个问题，我看到报告中雁栖湖bpu参考的是UC Berkeley Sonic BOOM，请问南湖bpu参考的是什么呢？
> 
> 为什么一定要参考谁呢，我们可以独立设计。

好的，明白了，感谢你们的回复 :)

shinezyy, 2024-04-25:
> > 目前的对齐工作都是朝向昆明湖对齐的，BPU部分也是一样。 RTL的BPU设计在南湖到昆明湖的变化不是很大，因此和南湖也有一定相似程度。
> 
> 感谢回复，那我使用该仓库下gem5得到bpu的性能作为南湖bpu性能，这种做法是不是可取呢？

可以认为是一个近似，但是我们无法做出承诺。

具体地说，你在做研究的时候，用这个GEM5里的bpu来近似南湖的bpu，我觉得是ok的。

但是在更加正式的场合应该以RTL为准，例如面向商业产品的调研、面向官方的报告时，用这里的bpu来比，就不合适了。

kevinSun20, 2024-04-25:
> > > 目前的对齐工作都是朝向昆明湖对齐的，BPU部分也是一样。 RTL的BPU设计在南湖到昆明湖的变化不是很大，因此和南湖也有一定相似程度。
> > 
> > 
> > 感谢回复，那我使用该仓库下gem5得到bpu的性能作为南湖bpu性能，这种做法是不是可取呢？
> 
> 可以认为是一个近似，但是我们无法做出承诺。
> 
> 具体地说，你在做研究的时候，用这个GEM5里的bpu来近似南湖的bpu，我觉得是ok的。
> 
> 但是在更加正式的场合应该以RTL为准，例如面向商业产品的调研、面向官方的报告时，用这里的bpu来比，就不合适了。

好的，感谢 ：）

seanwang2046, 2026-01-06:
问题已解决