OpenXiangShan/GEM5 #52: gem5在fs模式下仿真使用16核CPU，CPU8-15无法上线

bunnybugs0826, 2024-06-05:
目前正在尝试用gem5做仿真，目标是想运行一个16核的CPU，在内核启动中遇到CPU8-15无法上线的问题
log如下
`[    0.004941] EFI services will not be available.
[    0.006221] smp: Bringing up secondary CPUs ...
[    1.026992] CPU8: failed to come online
[    2.053885] CPU9: failed to come online
[    3.080779] CPU10: failed to come online
[    4.107673] CPU11: failed to come online
[    5.134567] CPU12: failed to come online
[    6.161461] CPU13: failed to come online
[    7.188354] CPU14: failed to come online
[    8.215248] CPU15: failed to come online
[    8.215338] smp: Brought up 1 node, 8 CPUs`

OS启动完成后查看系统的信息如下：
`root@UCanLinux:~ # ls -lh /sys/devices/system/cpu/
total 0
drwxr-xr-x    3 root     root           0 Jan  1 00:00 cpu0
drwxr-xr-x    3 root     root           0 Jan  1 00:00 cpu1
drwxr-xr-x    2 root     root           0 Jan  1 00:00 cpu10
drwxr-xr-x    2 root     root           0 Jan  1 00:00 cpu11
drwxr-xr-x    2 root     root           0 Jan  1 00:00 cpu12
drwxr-xr-x    2 root     root           0 Jan  1 00:00 cpu13
drwxr-xr-x    2 root     root           0 Jan  1 00:00 cpu14
drwxr-xr-x    2 root     root           0 Jan  1 00:00 cpu15
drwxr-xr-x    3 root     root           0 Jan  1 00:00 cpu2
drwxr-xr-x    3 root     root           0 Jan  1 00:00 cpu3
drwxr-xr-x    3 root     root           0 Jan  1 00:00 cpu4
drwxr-xr-x    3 root     root           0 Jan  1 00:00 cpu5
drwxr-xr-x    3 root     root           0 Jan  1 00:00 cpu6
drwxr-xr-x    3 root     root           0 Jan  1 00:00 cpu7
drwxr-xr-x    2 root     root           0 Jan  1 00:00 cpu8
drwxr-xr-x    2 root     root           0 Jan  1 00:00 cpu9
-r--r--r--    1 root     root        4.0K Jan  1 00:00 isolated
-r--r--r--    1 root     root        4.0K Jan  1 00:00 kernel_max
-r--r--r--    1 root     root        4.0K Jan  1 00:00 offline
-r--r--r--    1 root     root        4.0K Jan  1 00:00 online
-r--r--r--    1 root     root        4.0K Jan  1 00:00 possible
-r--r--r--    1 root     root        4.0K Jan  1 00:00 present
-rw-r--r--    1 root     root        4.0K Jan  1 00:00 uevent
root@UCanLinux:~ # cat /sys/devices/system/cpu/online
0-7
`
上线的CPU只有0-7，请问是否有使用gem5在fs模式下仿真使用16核CPU的经验，gem5本身对核数是不是有限制呢？

shinezyy, 2024-06-05:
可以用官方的 gem5 试一试，不排除我们改出问题来了。
目前版本的 xs-gem5 不计划支持多核，未来的版本会把 memory 相关修改切到 ruby 来实现较好的多核支持。

bunnybugs0826, 2024-06-06:
> 可以用官方的 gem5 试一试，不排除我们改出问题来了。 目前版本的 xs-gem5 不计划支持多核，未来的版本会把 memory 相关修改切到 ruby 来实现较好的多核支持。

使用官方的gem5 遇到了同样的问题。

bunnybugs0826, 2024-06-19:
> > 可以用官方的 gem5 试一试，不排除我们改出问题来了。 目前版本的 xs-gem5 不计划支持多核，未来的版本会把 memory 相关修改切到 ruby 来实现较好的多核支持。
> 
> 使用官方的gem5 遇到了同样的问题。

在官方gem5上使用opensbi可以正常使CPU8-15上线。之前上线失败的是使用了bbl

closed 2024-06-19 by bunnybugs0826