OpenXiangShan/GEM5 #53: what is the performance of current backport branch

XiaPZ, 2024-06-13:
I am trying to run some spec2006 benchmarks on xiangshan GEM5, currently I tried hmmer on backport branch, the weighted IPC is around 2.5 (spec2006 score is around 5.6 / GHz), that is far from 11 / GHz, so I am wondering what is the performance of current backport branch? Thanks

shinezyy, 2024-06-13:
Can you give more details on how you "tried" hmmer and how you relate IPC of hmmer agaist spec2006 score?

XiaPZ, 2024-06-13:
Sure, hmmer is built with gcc 11, O3 option, with rv64imafdc, score is calculated as (IPC * num_insts / 10^9 ) / ref_time

XiaPZ, 2024-06-13:
checkpoint is collected with NEMU, and run with util/warmup_scripts/simple_gem5.sh

shinezyy, 2024-06-13:
hmmer's performance is sensitive to B-extension, try gcc 12 with rv64gc_zba_zbb_zbc_zbs

XiaPZ, 2024-06-13:
After recompiled hmmer with B-extension, now the IPC is around 4.9,  you are right, hmmer is sensitive to B-extension.

closed 2024-06-13 by XiaPZ