OpenXiangShan/GEM5 #174: How to contribute simple code to xiangshan gem5

zybzzz, 2024-09-29:
I wrote a simple shell script to collect nix generated checkpoints from [this](https://github.com/xieby1/deterministic_checkpoints) , run them on gem5 and get the results, analyze the results [via topdown](https://github.com/shinezyy/gem5_data_proc) and output them.

It's a simple piece of code that simply analyzes the data from the results it gets and prints the results in the terminal. I don't know if xiangshan GEM5 is willing to accept this kind of contribution. If so, what branch should I contribute to?

**It's just a simple shell script and not very useful, I just thought it might make it easier for newbies to get started.**

The result of the run is shown in the figure:
[image]

shinezyy, 2024-10-04:
I think it is similar to scripts in [gem5_data_proc](https://github.com/shinezyy/gem5_data_proc)? Would you like to contribute to gem5_data_proc?

zybzzz, 2024-10-08:
> I think it is similar to scripts in [gem5_data_proc](https://github.com/shinezyy/gem5_data_proc)? Would you like to contribute to gem5_data_proc?

Yes, this is similar to gem5_date_proc and actually uses batch.py. I would like to tidy up the code and contribute it to gem5_data_proc. thanks!

closed 2024-10-08 by zybzzz