OpenXiangShan/GEM5 #175: Changing fuPool in XiangshanCore has no impact on the simulation results

DCliuzhe, 2024-09-30:
I have replaced the fuPool in XiangshanCore with my own defined one(Actually with some changes in num of the units). But there is no difference in simulation results before and after the change. May I ask what is the reason?

shinezyy, 2024-10-04:
Can you provide config.ini file produced from both simulations ( before and after modifications)?

DCliuzhe, 2024-10-08:
Here they are.
[origin.txt](https://github.com/user-attachments/files/17286835/origin.txt)
[change.txt](https://github.com/user-attachments/files/17286836/change.txt)

tastynoob, 2024-10-08:
Actually, the fupool has been deprecated in xs-gem5, please see FUPool.py scheduler

shinezyy, 2024-10-08:
@DCliuzhe Can you also provide how you modify the code? For example, Screenshot or log of `git diff src`

DCliuzhe, 2024-10-08:
Of course. Actually, my change is simple:
diff --git a/src/cpu/o3/FuncUnitConfig.py b/src/cpu/o3/FuncUnitConfig.py
index f0f2d70772..73c5dc036f 100644
--- a/src/cpu/o3/FuncUnitConfig.py
+++ b/src/cpu/o3/FuncUnitConfig.py
@@ -130,7 +130,7 @@ class SIMD_Unit(FUDesc):
                OpDesc(opClass='VectorMisc'),
                OpDesc(opClass='VectorIntegerExtension'),
                OpDesc(opClass='VectorConfig')]
-    count = 2
+    count = 3
 
 class PredALU(FUDesc):
     opList = [ OpDesc(opClass='SimdPredAlu') ]

DCliuzhe, 2024-10-08:
My workload was compiled by using AM. The workload is a GEMM implemented in RVV.

eastonman, 2024-10-08:
I suppose your change just modifies the **default number** of Fu.
FUPOOL is no longer used in the current version of xs-gem5.
FYI I am recently modeling an Apple-M4-like core at https://github.com/OpenXiangShan/GEM5/commit/ecc98494ef940d7215921c4d828bc39afa53215c
Hopefully this is helpful to you.

DCliuzhe, 2024-10-08:
Thank you for the example you provided. Referring to your example, I redefined a Scheduler and called it in xiangshan.py. After increasing the parallelism of the program, I observed performance improvement.

```
class MyScheduler(Scheduler):
    IQs = [
        IssueQue(name='IQ_misc' , inoutPorts=1, size=1*24, fuType=[IntDiv()]),
        IssueQue(name='IQ_br', inoutPorts=2, size=2*24, fuType=[IntBRU()]),
        IssueQue(name='IQ_si', inoutPorts=2, size=2*24, fuType=[IntALU()]),
        IssueQue(name='IQ_ci', inoutPorts=2, size=2*24, fuType=[IntALU(), IntMult()]),
        IssueQue(name='IQ_stu', inoutPorts=2, size=2*24, fuType=[WritePort()]),
        IssueQue(name='IQ_ldu', inoutPorts=2, size=2*24, fuType=[ReadPort()]),
        IssueQue(name='IQ_cplx',inoutPorts=2, size=2*24,
            scheduleToExecDelay=3, fuType=[FP_MISC(), FP_SLOW(), FP_MAM(), FP_MAA()]),
        IssueQue(name='IQ_SIMD',inoutPorts=4, size=16*16, scheduleToExecDelay=3, fuType=[SIMD_Unit()])
    ]
    slotNum = 12
    xbarWakeup = True
```
```
if args.xiangshan_ecore:
    TestCPUClass = XiangshanECore(scheduler = MyScheduler())
    FutureClass = None
    args.cpu_clock = '2.4GHz'
else:
    TestCPUClass = XiangshanCore()
    FutureClass = None
```