OpenXiangShan/GEM5 #180: xs-dev分支在Ubuntu22.04上编译错误

SeriousAlpha, 2024-10-11:
**Describe the bug**

- What workload are you running? Baremetal workload, bbl+Linux+application or checkpoints?
- What is the expected behavior? What is the actual behavior?
- Can this issue be consistently reproduced? If so, what are the steps to reproduce? Can this issue be reproduced in latest xs-dev branch?
- Can you upload your image or checkpoint to somewhere to help us to reproduce the issue?

**To Reproduce**
Steps to reproduce the behavior:
- Build command: typically `PYTHON_CONFIG=/usr/bin/python3-config scons build/RISCV/gem5.opt --gold-linker -j`nproc` --install-hooks`

- The link to your image or checkpoint

**Expected behavior**
A clear and concise description of what you expected to happen.

**Error log or Screenshots**
 [SO Param] m5.objects.RubyTester, RubyTester -> RISCV/python/_m5/param_RubyTester.cc
 [SO Param] m5.objects.Process, EmulatedDriver -> RISCV/python/_m5/param_EmulatedDriver.cc
 [SO Param] m5.objects.AbstractMemory, AbstractMemory -> RISCV/params/AbstractMemory.hh
 [SO Param] m5.objects.PMAChecker, PMAChecker -> RISCV/python/_m5/param_PMAChecker.cc
 [     CXX] RISCV/mem/cache/prefetch/composite_with_worker.cc -> .o
 [     CXX] RISCV/python/_m5/param_PMAChecker.cc -> .o
 [     CXX] RISCV/mem/cache/prefetch/cmc.cc -> .o
 [     CXX] RISCV/mem/qos/mem_sink.cc -> .o
In file included from build/RISCV/mem/cache/prefetch/composite_with_worker.hh:6,
                 from build/RISCV/mem/cache/prefetch/composite_with_worker.cc:1:
build/RISCV/mem/cache/prefetch/cdp.hh: In destructor ‘virtual gem5::prefetch::CDP::~CDP()’:
build/RISCV/mem/cache/prefetch/cdp.hh:155:25: error: no matching function for call to ‘gem5::prefetch::CDP::~CDP()’
  155 |         Queued::~Queued();
      |                         ^
In file included from build/RISCV/mem/cache/prefetch/cdp.hh:53,
                 from build/RISCV/mem/cache/prefetch/composite_with_worker.hh:6,
                 from build/RISCV/mem/cache/prefetch/composite_with_worker.cc:1:
build/RISCV/mem/cache/prefetch/queued.hh:227:13: note: candidate: ‘virtual gem5::prefetch::Queued::~Queued()’
  227 |     virtual ~Queued();
      |             ^
build/RISCV/mem/cache/prefetch/queued.hh:227:13: note:   candidate expects 1 argument, 0 provided
 [     CXX] RISCV/mem/shared_memory_server.cc -> .o
 [     CXX] RISCV/cpu/testers/directedtest/InvalidateGenerator.cc -> .o
At global scope:
cc1plus: error: unrecognized command line option ‘-Wno-unused-private-field’ [-Werror]
cc1plus: all warnings being treated as errors
scons: *** [build/RISCV/mem/cache/prefetch/composite_with_worker.o] Error 1
scons: building terminated because of errors.
*** Summary of Warnings ***
Warning: Deprecated namespaces are not supported by this compiler.
         Please make sure to check the mailing list for deprecation announcements.
Warning: Couldn't find HDF5 C++ libraries. Disabling HDF5 support.

[image]

**Necessary information on versions**
- XS-GEM5 version: [e.g. xs-dev branch commit f9dc472571de4b6ab83ea692f30e5ba454adf0f2 (HEAD -> xs-dev, origin/xs-dev, origin/HEAD)]
- NEMU version used as reference design: [e.g. master branch or commid: deadbeaf]
- Checkpoint restorer version used: [e.g. main branch of OpenXiangShan/LibCheckpointAlpha, or commit: deadbeaf of xxx]

**Additional information**
If you generate checkpoints your self, you can optionally provide version information of following components:
- riscv-linux
- riscv-rootfs
    - initramfs.txt
- riscv-pk or OpenSBI
- GCPT restorer (checkpoint restorer), it might be NEMU in older versions or OpenXiangShan/LibCheckpointAlpha or OpenXiangShan/LibCheckpoint in newer versions

**Additional context**
Add any other context about the problem here.

tastynoob, 2024-10-11:
compiler version？
It shouldn't report “cc1plus: error: unrecognized command line option ‘-Wno-unused-private-field’ [-Werror]”
How about using clang++

SeriousAlpha, 2024-10-11:
> compiler version？ It shouldn't report “cc1plus: error: unrecognized command line option ‘-Wno-unused-private-field’ [-Werror]” How about using clang++

alpha@DESKTOP-16U29FL:~/project/XSGEM5$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.4.0-1ubuntu1~20.04.2' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-9QDOt0/gcc-9-9.4.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2)

tastynoob, 2024-10-11:
Do you try to use clang++?
We use g++=11.4.0
Or maybe you can remove this argument and
remove code line "Queued::~Queued();" in file cdp.hh:155

SeriousAlpha, 2024-10-11:
> "Queued::~Queued();"

"Queued::~Queued();" should be remove,otherwise,use gcc 9.4.0，it will occurr compilation errors

closed 2026-01-06 by seanwang2046