OpenXiangShan/GEM5 #181: Development Roadmap

jensen-yan, 2024-10-11:
Here is the development roadmap. Contributions and feedback are welcome.
# Project Roadmap

## 1. Dynamic Vector Length Support

- [ ] Implement variable vector length (not fixed at 128).

## 2. Value Prediction

- [ ] Loop iterators.
- [ ] vl prediction.
- [ ] Generic value prediction.

## 3. Speculative Memory Bypassing

- [ ] Table-less implementation.
- [ ] Dynamic bypassing.

## 4. Top-down Analysis Framework

- [ ] Further breakdown of details in top-down approach.
- [ ] Detailed documentation for top-down methodology.
- [ ] TIP/TEA integration.

## 5. Front-End Optimizations

- [ ] 8-wide instruction issue
- [ ] Two-taken branch predictor
- [ ] uBTB: Expand for improved timing guarantees.
- [ ] TAGE:
  - [ ] Increase from 5-table to 10-table cascaded comparison for better timing and reduced MPKI.
  - [ ] Compare increasing the number of levels versus increasing the number of entries.
- [ ] Fetch Pipeline:
  - [ ] Refactor for two-taken + two-fetch mechanism.
  - [ ] Enable single-cycle access to multiple cache lines.
  - [ ] Clean up fetch and cache pipeline code.

## 6. Back-End Improvements

- [ ] Load/Store calibration: challenging, important, but not urgent.
- [ ] Functional unit (FU) calibration.

zybzzz, 2024-10-13:
I am excited to see the RoadMap, and the toolchain provided by Xiangshan has significantly improved the execution speed of gem5 and shortened the debugging time.

I'm a current master's student in computer architecture, and my current work is focused on **value prediction**, but I'm also interested in top-down analysis methods. I'd be happy to contribute to xs-gem5, but considering that xs-gem5 is in the internal development phase, I am unsure whether it is open to contributions from ordinary developers in the open-source community.

If so, could I send an email to the appropriate people? I'll describe my background, what I can do, and what I'd like to do in the email.

@jensen-yan

shinezyy, 2024-10-13:
> I am excited to see the RoadMap, and the toolchain provided by Xiangshan has significantly improved the execution speed of gem5 and shortened the debugging time.
>
> I'm a current master's student in [...]

xs-gem5 has ended internal development phase, and is now open to contributions!

On contributing to topdown/value prediction, we can open another issue to discuss. And you can also e-mail me (zhouyaoyang@bosc.ac.cn or archshinez@outlook.com) and @jensen-yan

muke101, 2026-02-28:
Hi, this is really cool and I hope a lot of progress can be made in implementing these more advanced techniques.

I've Implemented a newer MDP algorithm called PHAST (https://webs.um.es/aros/papers/pdfs/sskim-hpca24.pdf) based on another Gem5 fork. It achieves much higher accuracy than Store Sets (the open source version at least. I see this repo implements a better version that I'm keen to evaluate PHAST against too). It feels it'd be appropriate to aim to include the algorithm in this fork too. The repo is here: https://gitlab.com/muke101/gem5-phast

I wouldn't have time to integrate it to this repo myself, but I'd be happy to offer help to anyone who might want to take it on.