OpenXiangShan/GEM5 #184: gem5.opt 执行 checkpoint 时 Difftest failed

rmxpf, 2024-10-12:
环境描述：

GEM5：xs-dev f231aa596b60bb096a0892fbfc4d68dc713d000f
NEMU：master cf24515c85f5be898687959ab299ea276dbd7c56
DRAMsim3：master 29817593b3389f1337235d63cac515024ab8fd6e
[LibCheckpointAlpha](https://github.com/OpenXiangShan/LibCheckpointAlpha)：main c5c2fef74133fb2b8ef8642633f60e0996493f29
工具链：15.0.0（/nfs/home/hebo/TOOL/gnu-riscv64-toolchain/riscv-toolchain-gcc15-240613-noFFnoSeg2）

------

我遇到的问题：gem5.opt 执行 checkpoint 时 Difftest failed
我想要的：我想用最新的 GEM5 跑我的 checkpoint（gcb）

## 现象描述

在用 gem5.opt 执行 checkpoint 时 Difftest failed ，部分日志如下所示：

```bash
**** REAL SIMULATION ****
build/RISCV/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
build/RISCV/cpu/base.cc:1412: warn: Start memcpy to NEMU from 0x7fd92f9b8000, size=8589934592
build/RISCV/cpu/base.cc:1415: warn: Start regcpy to NEMU
build/RISCV/cpu/base.cc:1440: panic: Difftest failed!
Memory Usage: 17138044 KBytes
Program aborted at tick 61938
Aborted (core dumped)
```

感觉像是我的 difftest 启动失败了，我怀疑是我 riscv64-nemu-interpreter-so 或者 gcpt.bin 有问题，于是我尝试使用 GEM5/releases 里的编译好的文件：

[riscv64-nemu-interpreter-4332a525-so](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.2/riscv64-nemu-interpreter-4332a525-so) + 我自己的 gcpt.bin ，出现同样的现象

[nemu-gcbv-ref.so](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.0/nemu-gcbv-ref.so) + [gcb-restorer.bin](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.0/gcb-restorer.bin) ，出现同样的现象

[riscv64-nemu-interpreter-so](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.1/riscv64-nemu-interpreter-so) + [gcpt.bin](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.1/gcpt.bin) ，出现同样的现象

[nemu-gcbv-ref.so](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.0/nemu-gcbv-ref.so) + [gcb-restorer.bin](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.0/gcb-restorer.bin) ，出现同样的现象

[riscv64-nemu-interpreter-231008.so](https://github.com/OpenXiangShan/GEM5/releases/download/2023Oct27/riscv64-nemu-interpreter-231008.so) + [gcpt-restorer-231016.bin](https://github.com/OpenXiangShan/GEM5/releases/download/2023Oct27/gcpt-restorer-231016.bin) ，出现如下所示的现象，但还是没跑起来：

```bash
build/RISCV/mem/physical.cc:707: warn: Overriding Gcpt restorer
build/RISCV/mem/physical.cc:708: warn: gCptRestorerPath: /nfs/home/haokangda/hebo/TOOL/GEM5/GEM5-reuslt/test/gcpt-restorer-231016.bin
build/RISCV/mem/physical.cc:723: warn: Gcpt restorer file size 4352 is larger than limit 1792, is partially loaded
build/RISCV/mem/physical.cc:731: warn: gcpt restore size: 1792
build/RISCV/sim/system.cc:561: info: Restored from Xiangshan RISC-V Checkpoint
**** REAL SIMULATION ****
build/RISCV/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
build/RISCV/cpu/base.cc:1412: warn: Start memcpy to NEMU from 0x7fec8e83e000, size=8589934592
build/RISCV/cpu/base.cc:1415: warn: Start regcpy to NEMU
address (0x0000000010000000) is out of bound {???} [0x0000000000000000, 0x0000000000000000] at pc = 0x0000000010000000
  $0: 0x0000000000000000   ra: 0x0000000000000000   sp: 0x0000000000000000   gp: 0x0000000000000000 
  tp: 0x0000000000000000   t0: 0x0000000000000000   t1: 0x0000000000000000   t2: 0x0000000000000000 
  s0: 0x0000000000080000   s1: 0x0000000000000000   a0: 0x0000000000000000   a1: 0x0000000000000000 
  a2: 0x0000000000000000   a3: 0x0000000000000000   a4: 0x0000000000000000   a5: 0x0000000000000000 
  a6: 0x0000000000000000   a7: 0x0000000000000000   s2: 0x0000000000000000   s3: 0x0000000000000000 
  s4: 0x0000000000000000   s5: 0x0000000000000000   s6: 0x0000000000000000   s7: 0x0000000000000000 
  s8: 0x0000000000000000   s9: 0x0000000000000000  s10: 0x0000000000000000  s11: 0x0000000000000000 
  t3: 0x0000000000000000   t4: 0x0000000000000000   t5: 0x0000000000000000   t6: 0x0000000000000000 
 ft0: 0x0000000000000000  ft1: 0x0000000000000000  ft2: 0x0000000000000000  ft3: 0x0000000000000000 
 ft4: 0x0000000000000000  ft5: 0x0000000000000000  ft6: 0x0000000000000000  ft7: 0x0000000000000000 
 fs0: 0x0000000000000000  fs1: 0x0000000000000000  fa0: 0x0000000000000000  fa1: 0x0000000000000000 
 fa2: 0x0000000000000000  fa3: 0x0000000000000000  fa4: 0x0000000000000000  fa5: 0x0000000000000000 
 fa6: 0x0000000000000000  fa7: 0x0000000000000000  fs2: 0x0000000000000000  fs3: 0x0000000000000000 
 fs4: 0x0000000000000000  fs5: 0x0000000000000000  fs6: 0x0000000000000000  fs7: 0x0000000000000000 
 fs8: 0x0000000000000000  fs9: 0x0000000000000000 fs10: 0x0000000000000000 fs11: 0x0000000000000000 
 ft8: 0x0000000000000000  ft9: 0x0000000000000000 ft10: 0x0000000000000000 ft11: 0x0000000000000000 
pc: 0x0000000010000000 mstatus: 0x0000000a00000000 mcause: 0x0000000000000000 mepc: 0x0000000000000000
                       sstatus: 0x0000000200000000 scause: 0x0000000000000000 sepc: 0x0000000000000000
satp: 0x0000000000000000
[... 131 lines cut ...]

## 额外的信息

这是我很久之前配置的 GEM5 跑同一个 checkpoint 文件，它是可以执行的。

执行命令：

```bash
/nfs/home/hebo/TOOL/GEM5/GEM5-internal/build/RISCV/gem5.opt /nfs/home/hebo/TOOL/GEM5/GEM5-internal/configs/example/fs.py \
--xiangshan-system --cpu-type=DerivO3CPU --mem-size=8GB --caches \
--cacheline_size=64 --l1i_size=64kB --l1i_assoc=8 --l1d_size=64kB \
--l1d_assoc=8 --l1d-hwp-type=XSCompositePrefetcher --short-stride-thres=0 \
--l2cache --l2_size=1MB --l2_assoc=8 --l3cache --l3_size=16MB --l3_assoc=16 \
--l1-to-l2-pf-hint --l2-hwp-type=WorkerPrefetcher --l2-to-l3-pf-hint \
--l3-hwp-type=WorkerPrefetcher --mem-type=DRAMsim3 \
--dramsim3-ini=/nfs/home/hebo/TOOL/GEM5/GEM5-internal/ext/dramsim3/xiangshan_configs/xiangshan_DDR4_8Gb_x8_3200_2ch.ini \
--bp-type=DecoupledBPUWithFTB --enable-loop-predictor \
--difftest-ref-so /nfs/home/hebo/TOOL/GEM5/NEMU-test/732e4ccd/NEMU/build/riscv64-nemu-interpreter-so \
--enable-difftest \
--generic-rv-cpt=/nfs/home/haokangda/hebo/BOSC/Simpoint_Checkpoint/auto_checkpoint/archive/archive/969cfff4cce9ca1ca51d45cbf8254e8d/checkpoint-0-0-0/astar_biglakes/57/_57_0.016584_.gz \
--gcpt-restorer=/nfs/home/hebo/TOOL/GEM5/NEMU-test/732e4ccd/NEMU/resource/gcpt_restore/build/gcpt.bin \
--warmup-insts-no-switch=20000000 --maxinsts=40000000
```

日志文件路径：/nfs/home/hebo/TOOL/GEM5/test/log.txt

shinezyy, 2024-10-12:
Is it related to RV-H @jueshiwenli ?

jueshiwenli, 2024-10-12:
can you show me the nemu config?

shinezyy, 2024-10-12:
> ```shell
> riscv64-gem5-ref_defconfig
> ```

He says he is using riscv64-gem5-ref_defconfig of cf24515c85f5be898687959ab299ea276dbd7c56

shinezyy, 2024-10-12:
> can you show me the nemu config?

He says he is using riscv64-gem5-ref_defconfig of cf24515c85f5be898687959ab299ea276dbd7c56

jueshiwenli, 2024-10-12:
> > can you show me the nemu config?
> 
> He says he is using riscv64-gem5-ref_defconfig of cf24515c85f5be898687959ab299ea276dbd7c56

the config need update, open RVH, i will update it in nemu later

tastynoob, 2024-10-14:
This looks more like checkpoint and gcpt restore was not loaded correctly
Try to use “[Jul 24, 2024 release](https://github.com/OpenXiangShan/GEM5/releases/tag/v1.0.0-beta.0)” this

rmxpf, 2024-10-14:
> This looks more like checkpoint and gcpt restore was not loaded correctly Try to use “[Jul 24, 2024 release](https://github.com/OpenXiangShan/GEM5/releases/tag/v1.0.0-beta.0)” this

谢谢你的回答，我根据你的描述尝试了 “[Jul 24, 2024 release](https://github.com/OpenXiangShan/GEM5/releases/tag/v1.0.0-beta.0)” 中的 "[gcb-restorer.bin](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.0/gcb-restorer.bin) + [nemu-gcbv-ref.so](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.0/nemu-gcbv-ref.so)" 和 “[gcbv-restorer.bin](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.0/gcbv-restorer.bin) + [nemu-gcbv-ref.so](https://github.com/OpenXiangShan/GEM5/releases/download/v1.0.0-beta.0/nemu-gcbv-ref.so)” 这两个组合，得到的还是同样的错误。我还对 [Releases](https://github.com/OpenXiangShan/GEM5/releases) 中已经编译好的其他文件进了尝试，具体可见上面的 “现象描述” 。之后我会更新 NEMU 仓库后重新尝试，希望能成功。