OpenXiangShan/GEM5 #200: gem5::RiscvISA::TLB::demapPageL2(gem5::Addr, uint64_t): Assertion `m_newEntry != nullptr' failed

zybzzz, 2024-11-07:
## question

A panic occurs when running checkpoints( [xieby1/deterministic_checkpoints@main](https://github.com/xieby1/deterministic_checkpoints/commits/main/) ) on a gem5 built with xs-dev, and the panic is supposed to be **tlb** related.

I'm using two versions of xs-dev(5acbc5335abede6972dc55bd85e5f3f56ee9d192, 1237bc41ee134544ca944686ff60f2b56e1ea8a2) and they both have the same problem, and it's an occasional problem that occurs with some checkpoints and not with others.

Throughout the execution of the program, I made sure that the checkpoints were correctly restored and that difftest never reported an error.

## env
**ref-model**: github release
**gcpt**:https://github.com/OpenXiangShan/LibCheckpointAlpha/commits/main/
**checkpoints**: generate by deterministic_checkpoints. In order to generate checkpoints quickly, I've only selected the test collection from spec2006, so the error above only occurs in one of the checkpoints in the test collection for now.

[checkpoint and log here](https://rec.ustc.edu.cn/share/09283c90-9cc3-11ef-8919-6f022a1d57aa)

## other
This is the configuration I used to generate the checkpoint, although it only generates a checkpoint for the **test** collection and the error occurs on a few checkpoints, but I think there might be a related error for the **ref** collection as well.
```nix
{
  simulator = "nemu";       # nemu or qemu
  intervals = 20000000;
  workload = "miao";
  size = "test";            # test or ref
  profiling_log = "profiling.log";
  checkpoint_log = "checkpoint.log";
  spec2006_path = /home/zybzzz/proj/openxiangshan/benchmarks/spec2006 ;
  checkpoint_format = "zstd";   # gz or zstd, qemu only support zstd compressed
}
```

zybzzz, 2024-11-07:
I'm willing to try to fix this, but not sure how soon ...

zybzzz, 2024-11-11:
This may be a tricky issue that is caused by **refreshing the tlb** when the **sfence.vma** command is executed. I found the following problems, listed below, and given my limited knowledge, I'm not sure if the problems I listed are accurate.

[image]

This is the instruction encoding of sfence.vma, and my runtime dprintf shows that panic occurs when **rs1 ! = x0 and rs2 ! = x0**.

The risc-v privileged instruction set version 1.13 specifies the following:

> If rs1≠x0 and rs2≠x0, the fence orders only reads and writes made to leaf page table entries corresponding to the virtual address in rs1, for the address space identified by integer register rs2. Accesses to global mappings are not ordered. The fence also invalidates all address-translation cache entries that contain leaf page table entries corresponding to the virtual address in rs1 and that match the address space identified by integer register rs2, except for entries containing global mappings.

The code to flush the tlb is implemented in the **demapPage** and  **demapPageL2** method in **arch/riscv/tlb.cc**, and it is in this function that I found the issue in question:

###  1.Multiple clears of l2tlb
https://github.com/OpenXiangShan/GEM5/blob/6b909efa383127bfb8db701608eb4073c0d56533/src/arch/riscv/tlb.cc#L798C1-L807C14

Actually I think it's enough that l2tlb only needs to demap once in this process.

###  2.All entries in l2tlb are cleared, not entries pointing to the last level of the page table.
https://github.com/OpenXiangShan/GEM5/blob/6b909efa383127bfb8db701608eb4073c0d56533/src/arch/riscv/tlb.cc#L865C5-L897C6

The **if** here causes all l2tlb items to be cleared, which is against the documentation. It is also the fact that all of them are cleared that causes the following code to make another attempt at clearing the cleared tlb, resulting in a panic.

https://github.com/OpenXiangShan/GEM5/blob/6b909efa383127bfb8db701608eb4073c0d56533/src/arch/riscv/tlb.cc#L866C1-L874C10

The **global tlb entry** should not be invalidated in this case either.

---

**These are the issues I've found for now, I'm not sure if there are other issues in the tlb related implementation.** I also think the tlb related code needs **some additional comments** as well as **some renaming of variables**, this part of the code looks a bit hard to understand.

I'm not sure if my understanding is right, and if it is, I'm willing to raise pr to fix this in the future.

@shinezyy

jueshiwenli, 2024-11-14:
fixed