OpenXiangShan/GEM5 #234: store unaligned fault not save in inst->fault

zybzzz, 2024-12-16:
## question
We can see that in ```lsq.cc pushRequest```, the program tries to handle the unaligned store when the inserted request is a store request.

https://github.com/OpenXiangShan/GEM5/blob/38a5253405119e5e63d3cb6e7fad90b691fa111d/src/cpu/o3/lsq.cc#L918C1-L924C6

Here the program returns the fault as soon as it detects an unaligned store memory access.

https://github.com/OpenXiangShan/GEM5/blob/38a5253405119e5e63d3cb6e7fad90b691fa111d/src/cpu/o3/lsq_unit.cc#L1021C1-L1037C6

After receiving this exception, ```lsq_unit.cc executeStore``` still returns the exception to the calling function.

https://github.com/OpenXiangShan/GEM5/blob/38a5253405119e5e63d3cb6e7fad90b691fa111d/src/cpu/o3/iew.cc#L1425C15-L1448C18

Unfortunately, in ```iew.cc```, **the fault is not saved to inst->fault and is sent to the commit stage, which will cause an error once the store is committed.**

## analyze

Maybe there is a point to the design itself, but I just don't understand the code and what it means, and then this issue of mine will be meaningless.（有可能是我没读懂代码，这部分的设计本身就是这样）

---
But if it wasn't, I would think that this error would have been detected early on, since **we have difftest turned on at all times, but the reality is that difftest has never reported a problem due to this error**, which is confusing to me.

So I picked at least 4 checkpoints to analyze their logs and summarized the following points:
1.  Store unaligned faults are inherently rare, and in most checkpoints no such faults are generated, so difftest reports no errors.
2.  In the only checkpoint that produced store unaligned faults, each of the unaligned faults was due to the pipeline squash of the previous command **causing the store not to be finalized for commit**, so difftest did not report the problem.

---

If this is indeed a bug, I think a better solution would be to move the unaligned store judgment to the, ```lsqunit.cc write``` method, as is the case with the unaligned load judgment, and be careful to save the fault to inst.

happy-lx, 2024-12-16:
I encountered the same problem too when reading the code before.

[image]
I observed that the function `initiateAcc` would set fault to the DynInst which later calls `pushRequest` in lsq.cc.

tastynoob, 2024-12-16:
If store has page fault or others, it will set fault at PushRequest.
[image]

zybzzz, 2024-12-16:
> I encountered the same problem too when reading the code before. [...]

You are correct, I didn't notice that when I was reading the code. Thanks. Actually the unaligned store is still saved by inst->fault.

zybzzz, 2024-12-16:
> If store has page fault or others, it will set fault at PushRequest. [...]

Thanks, I understand the code you pointed out on your side. But as @happy-lx said, unaligned stores are actually still saved and processed. xs-gem5 currently supports unalign store processing.

tastynoob, 2024-12-16:
Yes, I just found out store and load's make fault were distributed in different locations

zybzzz, 2024-12-16:
[image]

This part of the code seems fine for now, but moving it to ```write``` might be a better option. I haven't analyzed this situation yet. @tastynoob

closed 2024-12-16 by zybzzz