OpenXiangShan/GEM5 #318: Assertion failed "skidBuffer[tid].size() <= skidBufferMax"

DCliuzhe, 2025-03-10:
**Describe the bug**

I have run a simple GEMM without any optimization on XSGEM5. The GEMM Kernel is a bare-metal binary file compiled using AM. And the GEM5 reports an assertion failed "skidBuffer[tid].size() <= skidBufferMax".

**To Reproduce**

```
./build/RISCV/gem5.opt configs/example/xiangshan.py --generic-rv-cpt ~/nexus-am/apps/gemm/build/gemm_basic-riscv64-xs.bin --raw-cpt
```
My source code of the kernel is:
```
#include <stdlib.h>

#define M 64  
#define K 128   
#define N 64   

void init_matrix(float* A, float* B, float* C) {

    for (int i = 0; i < M; i++) {
        for (int j = 0; j < K; j++) {
            A[i * K + j] = 1.0f;
        }
    }

    for (int i = 0; i < K; i++) {
        for (int j = 0; j < N; j++) {
            B[i * N + j] = 2.0f;
        }
    }

    for (int i = 0; i < M; i++) {
        for (int j = 0; j < N; j++) {
            C[i * N + j] = 0.0f;
        }
    }
}

void gemm(float* A, float* B, float* C) {
    for (int i = 0; i < M; i++) {
        for (int j = 0; j < N; j++) {
            for (int k = 0; k < K; k++) {
                C[i * N + j] += A[i * K + k] * B[k * N + j];
            }
        }
    }
}

int main() {
    float *A, *B, *C;
    A = (float *)malloc(M * K * sizeof(float));
    B = (float *)malloc(K * N * sizeof(float));
    C = (float *)malloc(M * N * sizeof(float));

    init_matrix(A, B, C);
    gemm(A, B, C);

    free(A);
    free(B); 
    free(C);

    return 0;
}
```

**Error log or Screenshots**

The detailed error log is:
```
gem5.opt: build/RISCV/cpu/o3/decode.cc:432: void gem5::o3::Decode::skidInsert(gem5::ThreadID): Assertion `skidBuffer[tid].size() <= skidBufferMax' failed.
Program aborted at tick 19558755
--- BEGIN LIBC BACKTRACE ---
./build/RISCV/gem5.opt(+0x1cbba10)[0x55ea16a05a10]
./build/RISCV/gem5.opt(+0x1cde8ac)[0x55ea16a288ac]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x14420)[0x7f1c44c5b420]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb)[0x7f1c43b4f00b]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b)[0x7f1c43b2e859]
/lib/x86_64-linux-gnu/libc.so.6(+0x22729)[0x7f1c43b2e729]
/lib/x86_64-linux-gnu/libc.so.6(+0x33fd6)[0x7f1c43b3ffd6]
./build/RISCV/gem5.opt(+0xcc7531)[0x55ea15a11531]
./build/RISCV/gem5.opt(+0xcc7680)[0x55ea15a11680]
./build/RISCV/gem5.opt(+0xcc7a19)[0x55ea15a11a19]
./build/RISCV/gem5.opt(+0xccaa07)[0x55ea15a14a07]
./build/RISCV/gem5.opt(+0xcad8e6)[0x55ea159f78e6]
./build/RISCV/gem5.opt(+0x1cceea2)[0x55ea16a18ea2]
./build/RISCV/gem5.opt(+0x1cf6e24)[0x55ea16a40e24]
./build/RISCV/gem5.opt(+0x1cf7bfe)[0x55ea16a41bfe]
./build/RISCV/gem5.opt(+0x1c31b80)[0x55ea1697bb80]
./build/RISCV/gem5.opt(+0x88c474)[0x55ea155d6474]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(+0x144ace)[0x7f1c44dceace]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(_PyObject_Call+0x6f)[0x7f1c44dcdaaf]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x5c2b)[0x7f1c44dbc8db]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(+0x12be6a)[0x7f1c44db5e6a]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(_PyFunction_Vectorcall+0xdd)[0x7f1c44dc437d]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x4db7)[0x7f1c44dbba67]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(+0x12be6a)[0x7f1c44db5e6a]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(_PyFunction_Vectorcall+0xdd)[0x7f1c44dc437d]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x126a)[0x7f1c44db7f1a]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(+0x12be6a)[0x7f1c44db5e6a]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(_PyFunction_Vectorcall+0xdd)[0x7f1c44dc437d]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x4db7)[0x7f1c44dbba67]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(+0x12be6a)[0x7f1c44db5e6a]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(_PyEval_EvalCodeWithName+0x51)[0x7f1c44db5af1]
/data/anaconda3/envs/liuzhe/lib/libpython3.9.so.1.0(PyEval_EvalCodeEx+0x3a)[0x7f1c44db5a9a]
--- END LIBC BACKTRACE ---
Aborted (core dumped)
```

**Necessary information on versions**

- XS-GEM5 version: 45df8bf59bbea59127bf5bae454d523229563745
- NEMU version used as reference design:  No use of NEMU
- Checkpoint restorer version used: No use of Checkpoint restorer
- AM version: e0acf4609821b099d8e30cc62573024c43fe9001

**Additional context**
It seems like when the execution in loop grows complex, the assertion fails. If I define a temp variable to store the partial sum instead of assigning it directly to the ouput matrix, the kernel can run normally.

tastynoob, 2025-03-14:
You can try the newest xs-dev again, if still error, then try enlarge the skidBufferMax.

DCliuzhe, 2025-03-14:
I have tried the newest xs-dev and it ran normally. Thanks.

tastynoob, 2026-03-06:
We have now changed the forward tick of the XS gem5 pipeline (fetch.tick -> commit.tick) to a reverse tick(commit.tick -> fetch.tick), and the pipeline has been refactored. This problem no longer occurs.

closed 2026-03-06 by tastynoob

DCliuzhe, 2026-03-09:
Really appreciate the fix — thanks