OpenXiangShan/GEM5 #331: xs-gem5使用nemu生成的cpt时出现错误

duiek, 2025-03-13:
**一、关于nemu和cpt：**

> 1、nemu为master分支，make  riscv64-xs-cpt_defconfig配置。仅修改最大内存CONFIG_MSIZE
2、使用了nemu的sdcard支持，加载一个基于busybox的img
3、由于workload的需求，以rw的方式挂载ext4，bootargs = "root=/dev/mmcblk0p1 rootfstype=ext4 rw rootwait earlycon";

按照https://docs.xiangshan.cc/zh-cn/latest/tools/simpoint/ 的方法，制作了zstd格式的checkpoint，能够在nemu中恢复。

**二、关于xs-gem5：**

 1、使用xs-dev分支，scons build/RISCV/gem5.opt --gold-linker -j8编译

2、运行时指定了与nemu中CONFIG_MSIZE的内存大小以及nemu生成的cpt，命令如下：

> ./build/RISCV/gem5.opt  ./configs/example/xiangshan.py  --mem-size=128GB --generic-rv-cpt=/home/diann/xs-gem5/GEM5/NEMU/checkpoint_example_result/spec-cpt/gcpt/99/_99_0.070664_.zstd

3、在xiangshan.py中关闭了difftest，运行结果：

> **** REAL SIMULATION ****
build/RISCV/sim/simulate.cc:194: info: Entering event queue @ 0.  Starting simulation...
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr 2d272c 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/arch/riscv/pmp.cc:140: warn: pmp access fault.mode 2  vaddr ffffffe00001dcd8 
build/RISCV/cpu/o3/commit.cc:728: warn: cpu may be stucked
gem5 has encountered a segmentation fault!
段错误 (核心已转储)

[image]

请问是哪里出的问题？不好意思，刚刚接触还不太会用。我是否还需要指定生成cpt时用的img？

duiek, 2025-03-14:
使用initramfs然后制作checkpoint，在xs-gem5中恢复时也出现了错误，看起来是rob不能正常提交指令，这是为什么呢？

[image]

zybzzz, 2025-03-16:
我认为最好确定你在 nemu 中的镜像能不能在 gem5 中被恢复，gem5 对镜像的恢复是将 nemu 保存的内存镜像完全恢复到 gem5 中。你这里在 nemu 中加入了对 sdcard 的支持，你需要确认 sdcard 这一段能不能被恢复到 gem5 的内存中，这是第一点。第二，这个报错是 pmp 在检查对于内存访问的时候报的警告，你应该根据你的需求去检查这个组件对于你 sdcard 内存那段的支持情况。

---

I think it's best to first confirm whether the image in NEMU can be restored in gem5. In gem5, restoring an image means completely recovering the memory snapshot saved by NEMU into gem5. Here, you have added support for the SD card in NEMU, so you need to verify whether the SD card section can be restored into gem5's memory—this is the first point.

Second, the error you encountered is a warning issued by PMP when checking memory access. You should examine whether this component supports the SD card memory section based on your requirements.

以上是我的见解。 @duiek

zybzzz, 2025-03-16:
> 使用initramfs然后制作checkpoint，在xs-gem5中恢复时也出现了错误，看起来是rob不能正常提交指令，这是为什么呢？
> 
> [image]

Additionally, based on my experience, when encountering this situation, please enable the "Scheduler" and "Commit" DebugFlags, save the logs, and analyze them.

duiek, 2025-03-17:
关于initramfs的checkpoint，当我将nemu与gem5的内存大小设置为16GB时似乎不会出错，但是同一个workload在内存为128GB时就会出现cpu stucked的错误信息，除了DRAMsim3的ini文件，我还需要修改什么文件来配置内存吗？
使用--debug-flag=Commit,MMU,TLB,PageTableWalker，打印出的部分信息如下：

> 2625372: system.cpu.commit: [tid:0] [sn:988] Committing instruction with PC (0x80000880=>0x20004b3f56).(0=>1)
2625372: system.cpu.commit: [tid:0] [sn:988] Return Instruction Committed PC (0x80000880=>0x20004b3f56).(0=>1) 
2625372: system.cpu.commit: [tid:0] ROB has 0 insts & 320 free entries.
2625705: system.cpu.commit: Getting instructions from Rename stage.
2625705: system.cpu.commit: Trying to commit instructions in the ROB.
2625705: system.cpu.commit: [tid:0] ROB has 0 insts & 320 free entries.
2626038: system.cpu.commit: Getting instructions from Rename stage.
2626038: system.cpu.commit: Trying to commit instructions in the ROB.
2626038: system.cpu.commit: [tid:0] ROB has 0 insts & 320 free entries.
2626371: system.cpu.commit: Getting instructions from Rename stage.
2626371: system.cpu.commit: Trying to commit instructions in the ROB.
2626371: system.cpu.commit: [tid:0] ROB has 0 insts & 320 free entries.
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004b3f54
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2l1 key 0x2000000000
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004b8000
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2l1 key 0x2000000000
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004a8000
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2l1 key 0x2000000000
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004b3f54
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2l2
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004b8000
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2l2
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004a8000
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2l2
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004b3f54
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2l3
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004b8000
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2l3
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004a8000
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2l3
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004b3f54
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2sp1
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004b8000
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2sp1
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004a8000
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2sp1
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004b3f54
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2sp2
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004b8000
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2sp2
2626704: system.cpu.mmu.l2_shared: f_vpnl2l1 0x2000000000 f_vpnl2l2 0x2000400000 vpn 0x20004a8000
2626704: system.cpu.mmu.l2_shared: look up l2tlb in l2sp2
2626704: system.cpu.mmu.itb: miss in l1 tlb + l2 tlb
2626704: system.cpu.mmu.itb.walker: Starting page table walk for 0x20004b3f54
2626704: system.cpu.mmu.itb.walker: from_pre_req 0 f_level 2 from_l2tlb 0
2626704: system.cpu.mmu.itb.walker: Performing table walk for address 0x20004b3f54 shift 30 idx_f 0x80 ppn 0 satp.ppn 0xf3538c
2626704: system.cpu.mmu.itb.walker: Loading level2 PTE from 0xf3538c400 idx 0x80 idx_shift 0x2000000000 vaddr 0x20004b3f54
2626704: system.cpu.mmu.itb.walker:  sv39 size is 8
2626704: system.cpu.mmu.itb.walker: Sending packet 0xf3538c400 with sender state: 0x589d6300
2626704: system.cpu.mmu.itb.walker: Send read 0xf3538c400
2626704: system.cpu.mmu.itb: finish start
....................
2626704: system.cpu.mmu.itb.walker: Starting page table walk for 0x20004b3f80
2626704: system.cpu.mmu.itb.walker: from_pre_req 0 f_level 2 from_l2tlb 0
2626704: system.cpu.mmu.itb.walker: Coalescing walk for 0x20004b3f80(pc=0x20004b3f56) into 0x20004b3f54(pc=0x20004b3f56)
2626704: system.cpu.mmu.itb.walker: Walks in progress: 1. Coalesce req pc: 0x20004b3f56, addr: 0x20004b3f80 into currStates
2626704: system.cpu.mmu.itb: finish start
...................
2674989: system.cpu.mmu.itb.walker: Received timing response for sender state: 0x589d6300
2674989: system.cpu.mmu.itb.walker: Mark page fault for req 0x20004b3f54 (pc=0x20004b3f56).
2674989: system.cpu.mmu.itb: Create page fault #12 on 0x20004b3f56
2674989: system.cpu.mmu.itb.walker: Mark page fault for req 0x20004b3f80 (pc=0x20004b3f56).
2674989: system.cpu.mmu.itb: Create page fault #12 on 0x20004b3f56
2674989: system.cpu.mmu.itb.walker: req addr: 0x20004b3f80 main addr: 0x20004b3f54
2674989: system.cpu.mmu.itb.walker: Finished fault walk for 0x20004b3f54 (pc=0x20004b3f56)
2674989: system.cpu.mmu.itb: Create page fault #12 on 0x20004b3f56
2674989: system.cpu.mmu.itb.walker: Finished fault walk for 0x20004b3f80 (pc=0x20004b3f56)
2674989: system.cpu.mmu.itb: Create page fault #12 on 0x20004b3f56
[... cut ...] .walker: finish all walk return true
2674989: system.cpu.mmu.itb.walker: Walk complete for 0x20004b3f54 (pc=0x20004b3f56), erase it
................
2676987: system.cpu.commit: [tid:0] [sn:1049] Inserting PC (0x20004b3f56=>0x20004b3f5a).(0=>1) into ROB.
2676987: system.cpu.commit: Trying to commit instructions in the ROB.
2676987: system.cpu.commit: [tid:0] Can't commit, Instruction [sn:1049] PC (0x20004b3f56=>0x20004b3f5a).(0=>1) is head of ROB and not ready
2676987: system.cpu.commit: gem5 nop

duiek, 2025-03-17:
问题是否出在DRAMsim3的配置文件上？关于DRAMsim3似乎没有什么教程，不太理解其中的参数，如果我想模拟一个大小为128GB的DDR4内存，可以简单的修改channel_size以及channels吗？

zybzzz, 2025-03-18:
I am also unfamiliar with Dramsim3. You can try configuring it to 128GB first.

> 问题是否出在DRAMsim3的配置文件上？关于DRAMsim3似乎没有什么教程，不太理解其中的参数，如果我想模拟一个大小为128GB的DDR4内存，可以简单的修改channel_size以及channels吗？

closed 2025-03-20 by duiek