OpenXiangShan/GEM5 #338: What do the magic numbers 48 and 46 in the function buildKey in src/arch/riscv/tlb.cc mean?

yichenhit, 2025-03-17:
When I try to add SV48 support, there is a error about L2 TLB, the key of the problem is in the function buildKey. So I have some questions about this function :
1. What is the meaning of the magic numbers 48 and 46 in this function ? 
2. Is the key of L2 TLB in Xiangshan designed to be a fixed 52 bit ?

eastonman, 2025-03-17:
seems no special meaning, just concat ASID, translate mode and vpn into a tag

yichenhit, 2025-03-17:
Currently buildKey only uses VPN [0:45], which results in Sv48 VPN that may lose [46:47] 2 bits. If 46 or 48 doesn't mean anything, can I fix the bug by simply adding 2 to 48 and 46.

eastonman, 2025-03-17:
No, VPN in SV48 is not 46 bits, see official gem5 comments: https://github.com/gem5/gem5/blob/186a913a48f13bdd484fcdef17ac3e28d2b8b4c9/src/arch/riscv/tlb.cc#L68

yichenhit, 2025-03-17:
Oh, right, there is also a 12-bit offset in the VA. However, the input parameter vpn in the function buildKey does not seem to account for this offset, meaning its most significant bit remains the same as the corresponding bit in the VA. Therefore, in SV48 mode, VPN[3] will still lose the two highest bits.

eastonman, 2025-03-17:
Not familiar with tlb code, maybe @jueshiwenli can have a look

yichenhit, 2025-03-17:
> Oh, right, there is also a 12-bit offset in the VA. However, the input parameter vpn in the function buildKey does not seem to account for this offset, meaning its most significant bit remains the [...]

In Gem5, key[48:51] is ASID, and key[0:47] is VPN.
https://github.com/gem5/gem5/blob/186a913a48f13bdd484fcdef17ac3e28d2b8b4c9/src/arch/riscv/tlb.cc#L68
```
static Addr
buildKey(Addr vpn, uint16_t asid)
{
    return (static_cast<Addr>(asid) << 48) | vpn;
}
```

In Xiangshan Gem5，key[48:51] is ASID,  key[46: 47] is translation mode, key[0: 45] is VPN.
https://github.com/OpenXiangShan/GEM5/blob/xs-dev/src/arch/riscv/tlb.cc: 72
```
static Addr
buildKey(Addr vpn, uint16_t asid, uint8_t translateMode)
{
    return (static_cast<Addr>(asid) << 48) | (static_cast<Addr>(translateMode & 0x3) << 46) |
           (vpn & (((uint64_t)1 << 46) - 1));
}
```

It should be the two bits used by the translation mode that cause the SV48 problem.

jueshiwenli, 2025-03-17:
key[46: 47] is translation mode，Distinguish the four modes of translation after adding virtualization.

yichenhit, 2025-03-17:
Is the translation mode designed to use these two bits? I solved my problem by increasing the VPN size in Key by 2bit, i.e., using key[48:49] for translation mode and [50:53] for asid.

jueshiwenli, 2025-03-17:
> Is the translation mode designed to use these two bits? I solved my problem by increasing the VPN size in Key by 2bit, i.e., using key[48:49] for translation mode and [50:53] for asid.

The translation mode is these two, it seems correct.

yichenhit, 2025-03-21:
> > Is the translation mode designed to use these two bits? I solved my problem by increasing the VPN size in Key by 2bit, i.e., using key[48:49] for translation mode and [50:53] for asid.
> 
> The translation mode is these two, it seems correct.

Yes，It's right in SV39，by may not right in SV48。

jueshiwenli, 2025-03-21:
> > > Is the translation mode designed to use these two bits? I solved my problem by increasing the VPN size in Key by 2bit, i.e., using key[48:49] for translation mode and [50:53] for asid.
> > 
> > 
> > The translation mode is these two, it seems correct.
> 
> Yes，It's right in SV39，by may not right in SV48。

This design currently only considers SV39. SV48 will be incorporated later with potential refactoring.

yichenhit, 2025-03-21:
> This design currently only considers SV39. SV48 will be incorporated later with potential refactoring.
Ok，thanks，maybe you can check this out when you're supporting SV48. 👍

closed 2025-03-21 by yichenhit