OpenXiangShan/GEM5 #345: 请教一下nemu使用了sdcard加载img后制作的cpt，能在xs-gem5中使用吗？

duiek, 2025-03-20:
我希望使用nemu的sdcard功能加载img，但不确定这样制作的cpt能否在xs-gem5中使用？

我注意到xs-gem5中添加了NemuMMC设备，并提供了--mmc-img和--mmc-cptbin的选项。但使用上一直出现问题，我想知道：

1. 我目前使用时仅在命令中增加--mmc-img，指定生成cpt时使用的img。这能使xs-gem5与img进行io操作吗？

1. --mmc-cptbin选项似乎是用于恢复sdcard的状态？但我不太清楚该文件如何获得，nemu中生成cpt时能保存sdcard的状态吗？

1. 关于difftest，是否只有特定版本的nemu，如：gem-ref-main。才支持sdcard_difftest？

关于sdcard的功能，似乎使用较少，找不到太多说明。希望有熟悉的大哥们能解答我的疑惑，非常感谢！

duiek, 2025-03-24:
nemu以前的版本中似乎具有保存mmc-cptbin的功能？

我从gcpt-clint-sdcard 版本的nemu中提取了serialize_sdcard函数，用以保存mmc-cptbin文件。

在xs-gem5的xs-dev分支，git pull并重新编译后，当前版本：xs-dev e5b579a61eae2ddb726967a55e2e2fcd154dd034

之后指定mmc-img、mmc-cptbin运行zstd格式的cpt，出现如下问题：

[image]

有遇到类似情况的兄弟吗？？

tastynoob, 2025-03-27:
xs-gem5目前只支持内存镜像方式的cpt

closed 2025-09-13 by duiek