OpenXiangShan/GEM5 #356: FS sim for one cpt stucked from an over 20000 commit gap being detected

KKK10001, 2025-04-10:
全系统仿真
禁用Difftest
激励是NEMU针对gcc生成的切片中的一个，采用.zstd压缩格式存储

仿真运行了一小段时间，进入了Commit Stage的主函数，在一笔Change-Id: Ie38d1839494561921006e79dfd63b661546c862d的提交中添加了对当前的时刻与前一次提交的时刻的间隔的assert:

[image]

如下监测所示：显然当前确实距离前一次提交的间隔超过了预设的阈值
[image]

模拟器的代码逻辑暂时没有任何改动
从现象上看就是由于上述限定的逻辑本身引起的

那么，我有两个疑问：
1. 这一设限是为了揭露提交吞吐率不满足期望而有意为之？
2. 针对这个写死的逻辑，一般的回归测试比如香山的Daily/Weekly Regression的通过是基于怎样的配置？目前我用作测试的切片生成过程中的聚类算法所采用的K-means算法以及投影算法都是随机种子-seedkm ${random1} -seedproj ${random2}。我相信这个应该不会要求使用者写死

eastonman, 2025-04-10:
No instruction committed for 20,000 cycles is likely a simulator bug if the workload is correct. If the ckpt is **incorrect**, it may incur an infinite exception loop, which will stall the CPU, and no more instructions can be executed.

Recommend first try restoring your ckpt on NEMU. If your ckpt runs on NEMU, then try difftesting with NEMU.

KKK10001, 2025-04-10:
> No instruction committed for 20,000 cycles is likely a simulator bug if the workload is correct. If the ckpt is **incorrect**, it may incur an infinite exception loop, which will stall the CPU, and [...]

我第一步就是用NEMU restore cpt，是可以跑通的（参照视频，从terminal看是正常退出）
补充说明：跟视频的推荐配置不同的点在于我此时设定的interval=100000000（1亿条指令），预热是10000000（1千万条），视频推荐设置interval比我当前小了一个量级。

[image]

不过才跑了这么点指令就出错，应该还没到interval设多设少的地步

我可以试一下以NEMU作为ref之后的现象

KKK10001, 2025-04-11:
上面的NEMU第一条指令提交之后跑飞是由于我make配置错了，应该是make riscv64-gem5-ref_defconfig而非README.md中的riscv64-xs-ref_defconfig，这个配置应该针对的是香山的RTL

重新编译生成SO并联调。目前看到的trace.out最终的片段如下所示：

[image]

列举的比对项都匹配上了，看起来纯粹是因为XS-GEM5对PC=0x80000880这条指令解码成了MRET

KKK10001, 2025-05-10:
谢谢

closed 2025-05-10 by KKK10001